lxqt/lxqt-powermanagement #169: Not showing UPS as battery

tenplus1, 2019-08-14:
Would is be possible to have lxqt powermanagement show my ups as a battery like it does in xfce, gnome and kde ?

dev-zero, 2020-03-30:
... and maybe also the batteries of devices connected via bluetooth (the ones I see with `upower --dump`)?

adjamhub, 2021-01-15:
In my opinion, adding support for UPS is just 1 line of code. But I don't have one. Is there anyone willing to test it for me?

adjamhub, 2021-01-15:
about showing other kind of batteries... I can probably add them in the "information" widget. Showing in the systray is probably exaggerated. Isn't it?

tenplus1, 2021-11-09:
It seems no other tools are needed to check the ups battery (apcupsd or nuts), by typing 'upower -d' into terminal it shows all relevant ups battery information for use:
```
320ms2h:~$ upower -d
Device: /org/freedesktop/UPower/devices/ups_hiddev0
  native-path:          /sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-3/3-3:1.0/usbmisc/hiddev0                                                                                                    
  vendor:               APC                                                                            
  model:                870.O3 .I                                                                      
  serial:               5Q1402T02229                                                                   
  power supply:         yes                                                                            
  updated:              Tue 09 Nov 2021 08:22:34 GMT (15 seconds ago)                                  
  has history:          yes                                                                            
  has statistics:       yes                                                                            
  ups                                                                                                  
    present:             yes                                                                           
    state:               fully-charged                                                                 
    warning-level:       none                                                                          
    time to empty:       31.6 minutes                                                                  
    percentage:          100%                                                                          
    icon-name:          'battery-full-charged-symbolic'                                                
                                                                                                       
Device: /org/freedesktop/UPower/devices/DisplayDevice                                                  
  power supply:         yes                                                                            
  updated:              Tue 09 Nov 2021 08:18:04 GMT (285 seconds ago)                                 
  has history:          no                                                                             
  has statistics:       no                                                                             
  ups                                                                                                  
    present:             yes
    state:               fully-charged
    warning-level:       none
    time to empty:       31.6 minutes
    percentage:          100%
    icon-name:          'battery-full-charged-symbolic'

Daemon:
  daemon-version:  0.99.13
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  no
  critical-action: HybridSleep