lxqt/lxqt-powermanagement #178: lxqt-powermanagement segfault and doublefloat

fedya, 2019-10-26:
i'm trying to launch lxqt on Raspberry pi3b+ and got weird messages from lxqt-powermanagement binary

`lxqt-powermanagement: /builddir/build/BUILD/double-conversion-3.1.5/double-conversion/double-conversion.cc:371: static void double_conversion::DoubleToStringConverter::DoubleToAscii(double, double_conversion::DoubleToStringConverter::DtoaMode, int, char *, int, bool *, int *, int *): Assertion `!Double(v).IsSpecial()' failed.
`

gdb reporting me this

```
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /usr/bin/lxqt-powermanagement 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ff1eb6110 (LWP 5932)]
[New Thread 0x7ff0ab6110 (LWP 5935)]
lxqt-powermanagement: /builddir/build/BUILD/double-conversion-3.1.5/double-conversion/double-conversion.cc:371: static void double_conversion::DoubleToStringConverter::DoubleToAscii(double, double_conversion::DoubleToStringConverter::DtoaMode, int, char *, int, bool *, int *, int *): Assertion `!Double(v).IsSpecial()' failed.

Thread 1 "lxqt-powermanag" received signal SIGABRT, Aborted.
0x0000007ff678e338 in raise () from /lib64/libc.so.6
(gdb) bt
#0  0x0000007ff678e338 in raise () from /lib64/libc.so.6
#1  0x0000007ff6774f9c in abort () from /lib64/libc.so.6
#2  0x0000007ff6782c9c in __assert_fail_base () from /lib64/libc.so.6
#3  0x0000007ff6782d00 in __assert_fail () from /lib64/libc.so.6
#4  0x0000007ff60bb3d4 in double_conversion::DoubleToStringConverter::DoubleToAscii(double, double_conversion::DoubleToStringConverter::DtoaMode, int, char*, int, bool*, int*, int*) () from /usr/lib64/libdouble-conversion.so.3
#5  0x0000007ff6e2fddc in ?? () from /usr/lib64/libQt5Core.so.5
#6  0x0000007ff70b53e8 in QTextStream::operator<<(double) () from /usr/lib64/libQt5Core.so.5
#7  0x00000000002145a0 in QDebug::operator<< (this=0x7fffffe780, t=nan(0x8000000000000))
    at /usr/include/qt5/QtCore/qdebug.h:158
#8  BatteryWatcher::batteryChanged (this=0x397330) at ../src/batterywatcher.cpp:99
#9  0x0000000000216b8c in PowerManagementd::settingsChanged (this=<optimized out>) at ../src/batterywatcher.cpp:71
#10 0x0000000000215c34 in PowerManagementd::PowerManagementd (this=0x7fffffe990) at ../src/powermanagementd.cpp:48
#11 main (argc=1, argv=<optimized out>) at ../src/main.cpp:60
(gdb) 

```

agaida, 2019-10-26:
and where this is a LXQt problem?

closed 2019-10-26 by agaida

berolinux, 2020-11-18:
Just ran into this away, and it does look like an LXQt problem.

BatteryWatcher::batteryChanged (this=0x396fb0) at ../src/batterywatcher.cpp:99

sends a NaN to a QTextStream.
This, in turn, calls double_conversion::DoubleToStringConverter::DoubleToAscii, which fails with an assert() if passed a NaN or infinity.

Arguably QTextStream should catch this and not call DoubleToAscii on NaN, but the likely problem is the fact that it's a NaN in the first place (division by zero when trying to figure out battery charge % on a box that doesn't have batteries, perhaps?)