lxqt/lxqt-powermanagement #188: Backlight change when computer is idle

selairi, 2020-04-06:
Hello, I have just written a simple tool to change the backlight of the display when the computer is idle:

[image]

This is in:
[https://github.com/selairi/lxqt-powermanagement](https://github.com/selairi/lxqt-powermanagement)

I think this could be useful for lxqt-powermanager, because it lets you save some battery of your laptop.

If there aren't suggestions, I will do a pull request.

Thanks.

tsujan, 2020-04-06:
Great! This is really needed.

selairi, 2020-04-06:
I'm think that is better with this plugin:
[https://github.com/lxqt/lxqt-panel/pull/1385](https://github.com/lxqt/lxqt-panel/pull/1385)

tsujan, 2020-04-06:
> I'm think that is better with this plugin

IMHO, its place is good but you've written the code and you know what's best for it.

selairi, 2020-04-10:
Sorry, but a I have a question:

I have been busy with my work and have been unable to work at LXQt. Now, I don't know if the process to merge pull request in the master branch has changed. Should I merge by myself my pull request (PR) in master branch when someone give me the "Good To Merge" ("GTM")?
When I sent a PR in the past, @agaida usually merged it. Has something changed? I have this message in [https://github.com/lxqt/lxqt-config/pull/579](https://github.com/lxqt/lxqt-config/pull/579):

>  @selari - please remove the conflicts and merge

Finally, the [https://lxqt.org/](https://lxqt.org/) doesn't work. Who is working on it?

Thanks.

tsujan, 2020-04-10:
> Should I merge by myself my pull request (PR) in master branch when someone give me the "Good To Merge" ("GTM")?

After a GTM, you could.

> When I sent a PR in the past, @agaida usually merged it. Has something changed?

@agaida hasn't been around for a while.

selairi, 2020-04-10:
@tsujan  Thanks

tsujan, 2020-04-15:
Done in https://github.com/lxqt/lxqt-powermanagement/pull/189

closed 2020-04-15 by tsujan