lxqt/lxqt-powermanagement #191: Please make it possible to change what the power button does

ElonSatoshi, 2020-04-17:
##### Expected Behavior

I'd like to be able to change whether pressing the power button will shut down, restart, suspend, or do nothing. 

##### Current Behavior

Currently there is no way (that I can find) of adjusting the behavior of the power button in LXQT's power management control panel.

##### Possible Solution

Add an option to adjust the behavior of the power button (Would it have a "power button watcher" checkmark?) to lxqt-config-powermanagement

##### Steps to Reproduce (for bugs)

1. Boot Lubuntu 19.10 from a USB
2. Go to settings and then power management
3. Try to find a way to adjust the behavior of the power button

##### Context

I'm trying out Lubuntu 19.10 on a little laptop that's slightly broken such that if you hold it the wrong way, or close/open the lid the wrong way, it triggers the power button, which shuts Lubuntu down every time unless I hack into /etc/systemd/logind.conf.

##### System Information

* Distribution & Version: Lubuntu 19.10 live
* Kernel: Linux lubuntu 5.3.0-18-generic
* Qt Version: `qmake: could not exec '/usr/lib/x86_64-linux-gnu/qt4/bin/qmake': No such file or directory`
* liblxqt Version: /usr/lib/x86_64-linux-gnu/liblxqt.so.0.14.1
* lxqt-build-tools Version: Not installed
* Package version: lxqt-powermanagement/eoan,now 0.14.1-0ubuntu1 amd64 [installed,automatic]

selairi, 2020-04-18:
I think that this feature should be in lxqt-powermanager.

tsujan, 2020-04-18:
> I think that this feature should be in lxqt-powermanager.

@selairi, please make a PR for it! Your knowledge and experience in this area is more than others'.

ElonSatoshi, 2020-04-20:
Thank you for your support, both of you!

selairi, 2020-04-20:
I have an alpha implementation in [https://github.com/selairi/lxqt-powermanagement](https://github.com/selairi/lxqt-powermanagement). 😊

It handles "suspend" action properly, but it can not "Turn off monitors".

[image]

I am not an artist and the stock icon should be changed. I have set the "sleep" stock icon.

I have to add the handle for suspend and hibernate keys. Some computers have a different key for each action.

tsujan, 2020-04-20:
> I have set the "sleep" stock icon.

We'll find a suitable icon later, when you make the PR.

selairi, 2020-04-24:
I have just done a PR: [https://github.com/lxqt/lxqt-powermanagement/pull/193](https://github.com/lxqt/lxqt-powermanagement/pull/193)

closed 2020-05-03 by selairi