lxqt/lxqt-powermanagement #201: compiler error on master

stdcerr, 2020-06-23:
##### Expected Behavior
Compiles without errors

##### Current Behavior
I created a dir `build/`, cd-ed into it, executed `cmake ..` followed by `make` and got:

##### Possible Solution

##### Steps to Reproduce (for bugs)
Checkout master at f640eb884a061b0af2d49540e3634e4e33cce1fa
1. create `build/` , cd into it
2. execute `cmake ..`
3. execute `make`
4. error will show up after the 44% mark:
/home/ron/src/lxqt/lxqt/lxqt-powermanagement/config/helpers.cpp: In function ‘void fillComboBox(QComboBox*)’:
/home/ron/src/lxqt/lxqt/lxqt-powermanagement/config/helpers.cpp:36:56: error: ‘PowerShowLeaveDialog’ is not a member of ‘LXQt::Power’
   36 |     comboBox->addItem(QObject::tr("Ask"), LXQt::Power::PowerShowLeaveDialog);
      |                                                        ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [config/CMakeFiles/lxqt-config-powermanagement.dir/build.make:308: config/CMakeFiles/lxqt-config-powermanagement.dir/helpers.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:242: config/CMakeFiles/lxqt-config-powermanagement.dir/all] Error 2
make: *** [Makefile:150: all] Error 2

##### Context
It won't let me compile the lxqt-powermanagement module from sources as I'm trying to compile and install lxqt. I have compiled 

##### System Information
* Distribution & Version: Arch
* Kernel: Linux-lts 5.4.46
* Qt Version: 5
* liblxqt Version: master HEAD 
* lxqt-build-tools Version: master HEAD
* Package version: NA

tsujan, 2020-06-23:
You can't compile a git component without first compiling and installing its latest git dependencies. If you want to compile and install LXQt from git, see https://github.com/lxqt/lxqt/wiki/Building-from-source#generalbuild-order.

`PowerShowLeaveDialog` is in git `liblxqt`.

closed 2020-06-23 by tsujan

stdcerr, 2020-06-23:
Hi tsujian,

That's what I have done (or so I thought), strange. I went back and recompiled & reinstalled **libxqt** and now the compilation & installation of **lxqt-powermanagement** completed fine, too! Strange but I'll take it, must have been a PEBCAK. 
Thanks a lot @tsujan !

tsujan, 2020-06-23:
> I went back and recompiled & reinstalled libxqt and now the compilation & installation of lxqt-powermanagement completed fine, too!

Good. Mistakes happen.