lxqt/lxqt-powermanagement #223: After removing battery not found anymore

stefonarch, 2021-01-01:
On my  old laptop (HP 655) the battery got unhooked and it crashed from suspension. 
After putting the battery in place and reboot powermanagement doesn't find it anymore, in no way, tried reboot with and without, always the message "battery not found, powersettings disabled" and no icon.
```
$ cat /sys/class/power_supply/BAT0/charge_now 
2393000
$ cat /sys/class/power_supply/BAT0/present 
1

```
Tried delete configuration, other kernel, new user - no effect.
                                       
* Distribution & Version: 32bit debian bullseye , recent github checkout

tsujan, 2021-01-01:
It may be an issue in `solid`. If you had a KDE installation besides LXQt, you could verify that — but installing KDE only because of this problem isn't worth the effort at all, especially on an old laptop. A KDE live ISO may not show the problem either.

Someone should check the code.

stefonarch, 2021-01-01:
Hm,  I installed falkon in the same time which did bring a lot of KDE dependencies with it. Will remove it and test.

tsujan, 2021-01-01:
Sorry, I should have been more explicit. LXQt uses `solid` to detect the battery. `solid` is a tire-1 KDE dependency and is used by KDE. Therefore, if KDE shows the same issue, the problem should be in `solid`, not in our code.

stefonarch, 2021-01-01:
Well, problem was sitting  in front of the screen: an autoremove not checked removed upower and libupower-glib3 in the same time.

closed 2021-01-01 by stefonarch