lxqt/lxqt-powermanagement #224: idle watcher should be disabled on fullscreen

adjamhub, 2021-01-04:
If you are watching a movie or doing something fullscreen, your idle time will increase very fast. I think it should be disabled.

##### Expected Behavior

idle watcher stops on fullscreen

##### Current Behavior

idle watcher works normally also on fullscreen

##### Possible Solution

I'm not sure about the best solution here. Add an option to set this behavior (enabled by default) or just implement it.
I can provide a patch to implement it when the right approach has been decided.

tsujan, 2021-01-04:
Good idea.

An option is needed, so that, when *the active window is in the fullscreen state*, the idle watcher is disabled and once the state or activeness is changed, it becomes enabled, provided that it was enabled in the first place. The default should be the current behavior (ignoring fullscreen+activeness).

Please don't use x11 codes when implementing it!

tsujan, 2021-01-04:
Another thing:

The option needs to work for the idle action and backlight change alike.

adjamhub, 2021-01-04:
ok!
I'll provide a patch about :+1:

closed 2021-01-08 by tsujan

binex-dsk, 2021-01-20:
I think it should also be disabled for video. I usually don't watch videos fullscreen, so this change doesn't help that whenever I'm watching videos the idle watcher kicks in. Similar to this one, maybe a toggle switch. Probably should be enabled by default

tsujan, 2021-01-20:
> I think it should also be disabled while watching video

There's no way of detecting video players. The video player itself should have a mechanism to show that it isn't idle (Qt video players can simply use `QCursor::setPos()` once in a while).

binex-dsk, 2021-01-20:
@tsujan I was thinking of playing videos on browser.

tsujan, 2021-01-20:
> I was talking of playing videos on browser.

Makes no difference.

PR #225 (2021-01-08): Add option (disabled by default) to disable idle watcher on fullscreen.
Fixes #224
I also removed deprecated KIdleTime::timeoutReached(int) method and
used the actual timeoutReached(int,int) instead.
commit 359f2b73 (2021-01-08): Add option (disabled by default) to disable idle watcher on fullscreen. (#225)

* Add option (disabled by default) to disable idle watcher on fullscreen.
Fixes #224
I also removed deprecated KIdleTime::timeoutReached(int) method and
used the actual timeoutReached(int,int) instead.

* Updating patch following Palinek suggestions

* Tsujan suggestions

* oops... forgot one ;)