lxqt/lxqt-powermanagement #230: Idle/lid watcher settings don't actually change anything, even after restart

binex-dsk, 2021-01-20:
##### Expected Behavior
Lid and idle watcher behavior should change when... they're changed

##### Current Behavior
I have to manually edit elogind configuration for it to work.

##### Possible Solution
I've got no clue. Could this maybe be expecting regular systemd-logind to be used, and tries changing it through systemd? If so, elogind support would be needed

##### Steps to Reproduce (for bugs)
1. Open lxqt-config-powermanagement
2. Change lid or idle watcher settings

##### Context
I wanted my configuration to have it suspend on lid close all the time except when connected to my monitor. In addition, I wanted to suspend after 20 minutes of idleness - mostly as a "band-aid fix" for the idle watcher still being enabled during video, which, by the way, is also something that I'd like changed.

##### System Information

* Distribution & Version: Artix Linux
* Kernel: 5.10.8.artix1-1
* Qt Version: 5.15.2-3
* liblxqt Version: 0.16.0-1
* lxqt-build-tools Version: not installed, lxqt was downloaded from the repositories (actually, I just installed Artix with the LXQt graphical installer)
* Package version: 0.16.0-1

rcastill, 2021-01-21:
Hey, I don't know if I relate to your exact problem, but take a look if the problem is lxqt-powermanagement or the lid detection event instead. I have a new laptop and my config does not work as expected. This project relies on upower notifying state changes. What I've been able to find is that the root of all problems is that the driver (probably kernel module "button") fails to notify the acpi event. 

You can debug using `upower --monitor-detail`. Also, systemd-logind ignores handling configuration at `/etc/systemd/logind.conf` if some other application like lxqt-powermanagement takes a hold of the corresponding systemd inhibitor. So, assuming you have the lid watcher enabled you should see something like this when running `systemd-inhibit`:

```bash
WHO                 UID  USER    PID  COMM            WHAT                 WHY                                       MODE 
...
UPower              0    root    1088 upowerd         sleep                Pause device polling                      delay
lxqt-powermanager   1000 user 1380 lxqt-powermanag handle-hibernate-key lxqt-powermanager controls hibernate key  block
lxqt-powermanager   1000 user 1380 lxqt-powermanag handle-power-key     lxqt-powermanager controls power key      block
lxqt-powermanager   1000 user 1380 lxqt-powermanag handle-suspend-key   lxqt-powermanager controls suspend key    block
lxqt-powermanagment 1000 user 1380 lxqt-powermanag handle-lid-switch    LidWatcher is in da house!                block
```
The last line means that systemd-logind gives up the control of lid state change handling to lxqt-powermanagement.

---

What to do if you have the same problem?

You can watch for the state of your lid with: `cat /proc/acpi/button/lid/LID0/state`. Polling lid state will:
1. Assure you that is not a hardware problem (if you see it changes when you open/close your lid)
2. If you poll it constantly, for some reason I don't understand, upower will notify of state changes, therefore lxqt-powermanagement will work as expected.

What I did:

`/usr/local/bin/lid-watcher.sh`
```
#!/bin/bash

while true; do
    cat /proc/acpi/button/lid/LID0/state
    sleep 2
done
```
The above is just a script that polls for your lid state every 2 seconds.

`$HOME/.config/autostart/lid-watcher.desktop`
```
[Desktop Entry]
Exec=lid-watcher.sh
Name=lid-watcher
Type=Application
Version=1.0
```
The above is the file you need to start the aforementioned script on lxqt session login (you can set it up from lxqt-config-session).

I hope this helps you or someone in the future.

binex-dsk, 2021-01-21:
@rcastill this problem is probably powermanagement as it also affects idle watcher.

Also, as I said in the issue I'm using elogind, but it's the same as systemd-logind so it should still apply.

Thanks though

tsujan, 2021-01-21:
The title was misleading; changed now.

tsujan, 2024-11-04:
According to GitHub, the original reporter is a "ghost" now. Closing...

Open another report if it's still a problem.

closed 2024-11-04 by tsujan