lxqt/lxqt-powermanagement #231: [Feature Request] Battery percentage

rcastill, 2021-01-21:
Can you consider adding an option for showing the battery percentage next to the icon? Maybe I could do a PR?

I found #175 and OP asked for remaining time + percentage. I understand remaining time is not possible, but battery percentage was not addressed by the response.

tsujan, 2021-01-21:
The tray icon shows the battery percentage.

closed 2021-01-21 by tsujan

rcastill, 2021-01-21:
I meant next to. What happens if the user chooses to use system icons instead of the custom one?

tsujan, 2021-01-21:
> What happens if the user chooses to use system icons instead of the custom one?

Then, he may want to use a good icon set compatible with Qt environments.

ahmadraniri1994, 2022-01-13:
Check this @rcastill , [Lxqt precentage icon.](https://www.reddit.com/r/Lubuntu/comments/ic1223/lubuntu_2004_battery_percent_indicator/)

stefonarch, 2022-01-13:
[image]

Or just use "custom command plugin":

` upower -i /org/freedesktop/UPower/devices/battery_BAT0|grep percent|awk {'print $2'}

`