lxqt/lxqt-powermanagement #232: [feature request] multiple actions

wxl, 2021-02-01:
It would be beneficial to allow multiple actions. Example: it would be nice to both lock and turn off the monitor upon idle. One simple way to do this might be to allow for a custom action to run and then one could just use a script.

dlm21, 2021-09-27:
I just used lxqt for the first time after installing the lxqt-kwin version of Garuda. The lack of this feature is something I'm having to hack around for the time being, but it would be much nicer to have the option to both turn off the monitors AND lock the screen, so when I sit back down at my desk and bump the mouse or whatever, I need to authenticate once the monitors come back on.

stefonarch, 2025-08-29:
An option "Turn monitor(s) off after mm:ss" in addition at the top? This would allow combining it.

"Execute script before action" would be the solution for geeks but in any way we shouldn't clutter it with too many options.

jonesmz, 2026-01-05:
For me the missing functionality would be the ability to set a different timer for

1. screen-off
2. suspend
3. hibernate

E.g.

screen-off at 15 minutes. Suspend at 30. Wake from suspend and hibernate to disk at 60.

stefonarch, 2026-01-05:
Something like this should be doable, moving out "Turn off monitor(s)" from the combobox dropwdown:

[image]

Waking up from sleep and hibernate could be quite tricky and fail on some hardware, not sure.

stefonarch, 2026-01-05:
With groupboxes:
[image]

They could be checkable - removing "do nothing" )but looks a little bit weird together with the overall checkable groupbox IMO:

[image]

closed 2026-01-22 by tsujan