lxqt/lxqt-powermanagement #238: Suspend loop after upgrade

slidinghotdog, 2021-02-25:
I was having some errors when compiling lxqt-panel then removed all lxqt packages and installed git versions from AUR, then after rebooting the system, first login it showed a lot of notifications about lxqt-powermanagement and suspended imediatelly. (all i could read from notifications was something about dbus, there were 10 or so notifications)

##### Expected Behavior

continue working as before

##### Current Behavior

suspends imediatelly after login (already fixed)

##### Possible Solution

switched to tty, removed `~./config/lxqt/lxqt-powermanagement.conf` and restarted sddm
##### Steps to Reproduce (for bugs)

Not sure if it will be reproducible but the previous setting was 100+ minutes idle to suspend (not sure of the values since i deleted the conf file)

##### Context

I already fixed on my system so no need for support, but there's probably some error when getting the values from older config file, possibly from the change in #228, and it could cause issues on next release.

##### System Information

* Distribution & Version: Archlinux
* Kernel: 5.11.1-arch1-1
* Qt Version: qt5-base 5.15.2-5
* liblxqt Version: 0.16.0.18.g0eb9dc7-1 (git version from AUR in feb 24)
* lxqt-build-tools Version: 0.8.0.6.g17d975c-1 (git version from AUR in feb 24)
* Package version: upgrading from 0.16.0-1 (repo version) to 0.16.0.20.g231759d-1 (git version from AUR in feb 24)

tsujan, 2021-02-26:
> possibly from the change in #228

It had a problem but it was fixed by the contributor, after @palinek told him about it.

With your report, it may need a second check.

slidinghotdog, 2021-03-01:
Tested on a fresh install on virtualbox and the same thing happened, updated section "Steps to Reproduce (for bugs)" in first post

tsujan, 2021-03-01:
Tomorrow I'll reread the suspicious commit. Thanks for your report and comments!

slidinghotdog, 2021-03-01:
Alright, added old config file so if using git version u should be able to test just replacing the file

tsujan, 2021-03-01:
@slidinghotdog 
Your guess was right. The problem is in https://github.com/lxqt/lxqt-powermanagement/pull/228: it reads the legacy setting incorrectly. I'll make a PR.

tsujan, 2021-03-01:
@slidinghotdog
A probable fix is here: https://github.com/lxqt/lxqt-powermanagement/pull/239

Could you please test it as you did before? If you do and it works, please also test a few idle times. I set the minimum to 1 minute because a smaller idle time wasn't practical for shutdown/suspend/hibernation.

slidinghotdog, 2021-03-01:
Ok, the fix works, tested a few different values and had no problems.

tsujan, 2021-03-01:
Thank you so much for everything you did here!

slidinghotdog, 2021-03-02:
No problem, thanks for the quick response!

closed 2021-03-08 by tsujan