lxqt/lxqt-powermanagement #253: Feature: Item in tray context menu to pause idle watcher

tsujan, 2021-04-18:
We have an option to disable the idle watcher when the active window is fullscreen. But, as @binex-dsk mentioned at https://github.com/lxqt/lxqt-powermanagement/issues/224#issuecomment-763846625, that's not enough. Opening the config dialog and disabling the idle watcher each time a video starts playing is not a solution because it isn't easy and also, the user may forget to re-enable it afterward.

That's the only reason I use `xset dpms` directly, instead of LXQt's idle watcher: Firefox and SMplayer (and maybe VLC, which I don't use), suppress screen blanking whenever a video is playing.

Actually, it's easy to suppress any idle watcher inside a Qt app but most developers don't know how or think in terms of difficult methods that depend on the environment. Anyway, many apps don't do it.

I'm working on adding a "Pause idle watcher" item to the context menu of the tray icon to test it. It has the following properties:

 1. It's enabled only when Idle Watcher is enabled.
 2. When it's checked, the Idle Watcher remains enabled but doesn't do anything (see 3 for the reason).
 3. Its state is not saved anywhere, so that if the user forgets about it, the idle watcher will be enabled with the next session.
 4. The user can't forget about it because a "pause" emblem will be shown on the tray icon if it's enabled.
 5. It's also disabled automatically whenever the config dialog is shown or closed.

Only 4 isn't done yet.

If you have a better method in mind, please tell me.

stefonarch, 2021-04-20:
It's for a long time I use this script:
https://github.com/stefonarch/toggle-idleness-watcher
It get's triggered when doing backup and update, but from script.

Thinking about it an "disable idleness watcher for 1-2 hour would be better.

tsujan, 2021-04-20:
> Thinking about it an "disable idleness watcher for 1-2 hours" would cover all needs.

One of my goals was not touching the configuration dialog. I'm using it and it works as intended. Your suggestion could be added to it later, in a separate patch.

> Btw this repo was unwatched for some reason

Using the button on the top right of the page:

[image]

stefonarch, 2021-04-20:
Yeah sure I did now, I remember that I activated all together somehow.
Screen of redshift-gtk, there are three durations:
[image]

tsujan, 2021-04-20:
OK, your screenshot convinced me. I'll add it before making the PR but with extra 3 and 4-hours items — some movies are long ;)

It might take a while (with testing)...

tsujan, 2021-04-21:
After I avoided 2 issues in `KIdleTime` and Qt itself, it became like this (I took the screenshot with theme icons instead of the default icon):

[image]

I think it's enough for all practical purposes, thanks to @stefonarch's idea.

stefonarch, 2021-04-21:
Well, credits go to redshift-gtk  ;)

tsujan, 2021-04-21:
I like practical ideas, especially when their implementations don't need lots of code changes  :)

binex-dsk, 2021-04-21:
if you like practical ideas then why not guacamole

closed 2021-07-22 by tsujan