lxqt/lxqt-powermanagement #254: Remove "Ask" option in Idleness watcher and lid watcher

stefonarch, 2021-04-20:
It doesn't make  sense asking the user what to do when his PC is idle or when laptop lid is closed,

* Distribution & Version: all, latest git

tsujan, 2021-04-20:
> It doesn't make sense asking....

You're completely right. We should remove it.

stefonarch, 2023-07-23:
Made a tentative with 2 different helpers but it doesn't work: the "ask" option is removed everywhere.
https://github.com/lxqt/lxqt-powermanagement/compare/master...stefonarch:lxqt-powermanagement:remove_ask_option?expand=1

tsujan, 2023-07-23:
The ask option doesn't make sense for the lid watcher. It may not be useful with the idleness watcher either, but I think it makes sense for the battery watcher.

closed 2023-08-20 by tsujan