lxqt/lxqt-powermanagement #255: The idleness fullscreen option can totally disable idle watcher

tsujan, 2021-04-21:
I don't know if the following problem was there from start or happened after an upgrade because I only recently started to use the idle watcher.

##### Steps to Reproduce

1. Set the idle time to one minute (to not waste your time) and set the action to "Turn off monitor(s)".
2. Make a window fullscreen and leave the computer for one minute and a few seconds. The  monitor will not be turned off, as it shouldn't.
3. Exit the fullscreen state.

Then nothing will happen after one, two, three,.. minutes. The idle watcher will be magically disabled — or at least it is here (confirmation would be appreciated).

##### Solution

It seems that if `KIdleTime::instance()->simulateUserActivity();` is called directly from inside the slot of `KIdleTime::timeoutReached` (as it is called in the code), no timeout will happen anymore.

This may be a bug in `KIdleTime` but the fix/workaround is easy: if `KIdleTime::instance()->simulateUserActivity();` is called by `QTimer::singleShot()`, the problem will disappear.

##### System Information

* Latest git LXQt

stefonarch, 2021-04-21:
It says "minimum 1.30", so maybe one minute is too fast? Will test later.

tsujan, 2021-04-21:
> so maybe one minute is too fast?

No, it isn't related to one minute. I suggested 1 minute only for a quick test. Please test and tell me the result!

Do you think we'll need a point release for this? If you confirm the issue and also its fix (I'll make a PR soon), making a point release will be easy.

stefonarch, 2021-04-21:
Can confirm the issue by now.

tsujan, 2021-04-21:
> Can confirm the issue by now.

Good.

And a point release?

stefonarch, 2021-04-21:
> And a point release?

Sure, it's a bug. But we could ship also the idleness thing at this point.

stefonarch, 2021-07-21:
It's some time I see our PC doesn't go in suspension always when idle.
I tracked down now the issue: idleness suspend works at session start, after one fullscreen of youtube and exit it doesn't work anymore.

tsujan, 2021-07-21:
Only suspension?

stefonarch, 2021-07-22:
> Only suspension?

Strange, looks like so, I switched to "lock screen" and it worked, switched back top suspend and it didn't.
Suspend itself works fine.

tsujan, 2021-07-22:
> Strange, looks like so.

It's strange, indeed. It shows that the problem isn't in idle watcher. I'll test with my old laptop. If I encounter the problem, I'll open another report because this one was about idle watcher.

tsujan, 2021-07-22:
I tested and everything was OK. What I did:

 1. Chose suspension as the action, set the timeout to 1 min., and marked "Disable idleness checks when fullscreen".
 2. Played a Youtube video in fullscreen with Firefox and didn't touch keyboard or mouse.
 3. Let the video be played for a little more than one minute. Nothing happened, as expected.
 4. Exited the fullscreen mode, stopped the video, closed Firefox, didn't touch keyboard or mouse, and waited one minute.

The computer suspended after one minute, as expected.

I close this again because what you see is definitely not related to the idle watcher but please open another report if you're sure that you have the latest git on the computer, where you did the test. Please also include every step, even those that may seem unimportant.

closed 2021-07-22 by tsujan

tsujan, 2021-07-22:
In my 2nd test, I let the video be played for 5 minutes before stopping it and exiting Youtube's fullscreen mode but I didn't closed Firefox afterward. Again, the laptop was suspended after one minute.

stefonarch, 2021-07-22:
In the meantime I tested on the laptop, both on AC and on battery, everything was fine.

tsujan, 2021-07-22:
Good.

So, your previous test happened on a desktop computer? Unfortunately, my desktop computer is too old and too slow; haven't turned it on for years and can't update anything on it.

stefonarch, 2021-07-22:
Back on the Desktop PC; restarted session. 
Unfortunately there are a lot of combinations now, with different AC and DC settings to test out.
> Unfortunately, my desktop computer is too old and too slow; haven't turned it on for years and can't update anything on it.

You could try to remove the battery for testing?

What I found so far;
* popup about no battery shows up and says it will disable, but the checkbox "Enable battery watcher" remains checked always (if you check it), so at every login there is this message, is a separate issue also confirmed in LUbuntu Vbox, will open it.
* I can trigger the suspend on idle not working by selecting "nothing" for Battery and "suspend" for AC.
* Looks like lockscreen works always

As said there are many combinations, will repeat and do more later.

tsujan, 2021-07-22:
> You could try to remove the battery for testing?

No, the battery is integrated into the case.

> but the checkbox "Enable battery watcher" remains checked

OK, but will the message also appear with login if you uncheck it?

tsujan, 2021-07-22:
In the end, someone (perhaps I) should read the code line by line and see whether the absence of a battery is treated incorrectly.

tsujan, 2021-07-22:
BTW, https://github.com/lxqt/lxqt-powermanagement/issues/261 is about lack of battery. But it's a feature request, not a bug report. For the bug you've probably encountered, please open a separate issue.

stefonarch, 2021-07-22:
> OK, but will the message also appear with login if you uncheck it?

No. 

I tried to repeat the idleness suspend issue on the old laptop with the battery removed, no success. So giving up this atm, at least I think I know by now which config works always on my desktop PC.