lxqt/lxqt-powermanagement #261: LXQt Power Management has problems without  battery

tsujan, 2021-04-22:
1. Battery idle time and backlight change on battery discharge are enabled without battery.
 2. There's no way to have a tray icon without battery. The tray icon is (or will be) useful for pausing idleness checks. It can also show that the power management exits.
 3. Does LXQt Power Management reacts to the removal or adding of battery?

tsujan, 2021-04-22:
This needs a lot of work:

 * A separate section is required for the tray icon in the settings dialog.
 * In that section, the icon preview should show the battery icons if there is a battery; otherwise it should show a power management icon. Since we have a theme option, a separate power management icon needs to be added to the code.
 * The tray icon should be removed from `batterywatcher.cpp` and, probably, inserted into `powermanagementd.cpp`. If there is no battery or if the battery watcher is disabled, the tray icon should be changed to the icon of power management (an internal icon or from the theme, depending on the settings).

I might start to work on it after https://github.com/lxqt/lxqt-powermanagement/pull/260 is merged.

stefonarch, 2021-04-22:
I think we should take some time to have a good approach. 
I was thinking about an "Laptop" section, where Lid and Battery monitor would go. It could even non be visible when there is no lid and no battery. Just an idea this.

tsujan, 2021-04-22:
> I think we should take some time to have a good approach.

The changes to the main code weren't as complex as I thought at first but I agree that we should find the best design in the settings code. That will require a discussion; it'll take time.

stefonarch, 2021-04-22:
I just tested, removing battery when laptop is plugged in AC doesn't trigger anything, reinserting it either,  restart session does disable battery monitor, restart powermanager probably too.