lxqt/lxqt-powermanagement #270: Not suspending on low battery

antis81, 2021-07-11:
Having set powermanagement to "suspend" mode on low battery. Timer message shows, but when it reaches 0 laptop keeps running. Log shows (Q)Dbus errors:

```log
Jul 11 13:34:31 nils-tux-man lxqt-powermanagement[1042]: BatteryChanged discharging: true chargeLevel: 49.3827 actionTime: QTime("13:34:32.133")
Jul 11 13:34:31 nils-tux-man lxqt-powermanagement[1042]: QString::arg: Argument missing: "Ruhezustand in 0 Sekunden" , 0
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: BatteryChanged discharging: true chargeLevel: 49.3827 actionTime: QTime("13:34:32.133")
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: QString::arg: Argument missing: "Ruhezustand in 0 Sekunden" , 0
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: BatteryChanged discharging: true chargeLevel: 49.3827 actionTime: QTime("13:34:32.133")
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: systemd: "CanHibernate" = "na"
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: ** Dbus error **************************
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: Error name  "org.freedesktop.DBus.Error.InvalidArgs"
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: Error msg   "Keine derartige Schnittstelle »org.freedesktop.DBus.Properties«"
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: ****************************************
Jul 11 13:34:32 nils-tux-man lxqt-powermanagement[1042]: dbusCall: QDBusInterface is invalid "org.freedesktop.ConsoleKit" "/org/freedesktop/ConsoleKit/Manager" "org.freedesktop.ConsoleKit.Manager" "CanHibernate"
Jul 11 13:34:43 nils-tux-man lxqt-powermanagement[1042]: onActivated 3
Jul 11 13:34:43 nils-tux-man lxqt-powermanagement[1042]: toggleShow
```

##### Expected Behavior
Desktop activates "hibernate/suspend" mode.

##### Current Behavior
Desktop keeps running.

##### Possible Solution
Fix/Update LXQt Dbus interface.

##### Steps to Reproduce (for bugs)
1. set lxqt-powermanagement to a certain level to trigger "hibernate/suspend" mode
1. plug cable and wait until the battery level is reached

##### Context
Common laptop setup.

##### System Information
* Distribution & Version: Manjaro 21.0.7 (Ornara)
* Kernel: 5.10 
* Qt Version: 5.15.2
* ~~liblxqt Version~~ 
* ~~lxqt-build-tools Version~~
* Package version: 0.17

tsujan, 2021-07-11:
The message seems to be about hibernation, not suspend. Can you hibernate your laptop from lxqt-leave (main menu)?

antis81, 2021-07-11:
In German translation it is a bit hard to differentiate those. However starting lxqt-leave reveals the "hibernate" feature is not available (at least systemd thinks so):

```log
❯ lxqt-leave
systemd: "CanSuspend" = "yes"
systemd: "CanHibernate" = "na"
** Dbus error **************************
Error name  "org.freedesktop.DBus.Error.InvalidArgs"
Error msg   "Keine derartige Schnittstelle »org.freedesktop.DBus.Properties«"
****************************************
dbusCall: QDBusInterface is invalid "org.freedesktop.ConsoleKit" "/org/freedesktop/ConsoleKit/Manager" "org.freedesktop.ConsoleKit.Manager" "CanHibernate"
```

Can lxqt-powermanagement filter out the feature selection then just like lxqt-leave does? In my case hibernate  ("Ruhezustand" in German) seems not available. Works fine with suspend ("Bereitschaft") on my system.

tsujan, 2021-07-11:
>  In my case hibernate ("Ruhezustand" in German) seems not available.

For hibernation, see https://wiki.archlinux.org/title/Swap and https://wiki.archlinux.org/title/Power_management/Suspend_and_hibernate.

This page seems to explain it with swap partition for Manjaro (your distro): https://www.vegard.net/manjaro-enable-hibernate/.

Strange that Manjaro doesn't have an automation tool for it! Or does it?

> Works fine with suspend ("Bereitschaft") on my system.

So, the problem isn't in power management. Closing (could be re-opened if a problem is reported with an *enabled* hibernation).

closed 2021-07-11 by tsujan

antis81, 2021-07-12:
It is still LXQt's powermanagement GUI/Tool. The option is available, but fails "hidden". The integration is incomplete at this point.

tsujan, 2021-07-12:
You mean disabling an option when it doesn't work? It seems like a valid feature request that needs another report. You could open one.