lxqt/lxqt-powermanagement #283: [Feature Request] Add custom command to run before suspend

georgebushdown, 2021-09-30:
### lxqt-config-powermanagement
[image]

I set up auto-suspend when the laptop power level reaches a level. I'm using slock to lock the screen on demand and would like it to be run before the auto-suspend.

Can you add a box for putting a custom command before suspending?

stefonarch, 2021-09-30:
It's not necessary for this: you can use xautolock and the setting in `Session Settings > lock screen before suspend`. I'm using i3lock with this setup.
https://github.com/lxqt/lxqt/wiki/ConfigScreensavers#other-screenlockers-using-xautolock

georgebushdown, 2021-09-30:
Oh I didn't know about this page, thanks.

georgebushdown, 2021-09-30:
I don't see any lxqt modules in the list.

stefonarch, 2021-09-30:
> I don't see any lxqt modules in the list. What could be the issue?

Sorry, I don't  understand: which list?  Why a module?

georgebushdown, 2021-10-01:
Under lxqt-config-sessions.

stefonarch, 2021-10-01:
No idea, never seen this. But it's not related to this feature request. 
My 2 cents; testing a new clean user, and if you can repeat the problem open an issue.

georgebushdown, 2021-10-01:
Not related? The wiki article tells to look in lxqt session settings, which is what I did.

stefonarch, 2021-10-01:
The modules should be listed sure, but you need this  option not a module:

[image]

georgebushdown, 2021-10-01:
[image]

Initially I did, but there was no effect. Turns out there was an issue with `xautolock`. I switched to `xss-lock` instead and it started working. Thanks.

closed 2021-10-01 by georgebushdown