lxqt/lxqt-powermanagement #295: Wrong tooltip and left click info in pending-charge state

stefonarch, 2022-01-18:
In rare circumstances (probably unplugging power supply for a short time, 5 min) the battery will not be charged immediately:

` upower -i /org/freedesktop/UPower/devices/battery_BAT`

```
 battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
```
In this state left click on the icon results in  "no battery" and tooltip says "Battery empty" too.
[upower manual](https://upower.freedesktop.org/docs/Device.html) mentions 2 states :
```
....
 5: Pending charge
6: Pending discharge
```

##### Expected Behavior

No wrong tooltip infos, and icon should display power adapter connected  when in fact it is.

##### Possible Solution

##### Steps to Reproduce (for bugs)

1.  Unplug the charger for 5 minutes
2. Plug it in and wait for updating upower until
3. + sign on icon disappears, tooltip will say "Empty (97%), left click says "Battery empty" too.
4. No emblem  for "adapter connected" but it should be displayed.

##### Context

Noticed only  testing in https://github.com/lxqt/lxqt-powermanagement/pull/294 , usually the power adapter is longer disconnected than some minutes. It looks like the battery has to go under ~95% to be recharged immediately or at all, at 97% it hangs always there.
##### System Information

* Distribution & Version: arch, latest git

palinek, 2022-01-18:
This is probably a misinterpretation of reported state from solid. This is in our code:
```cpp
QString BatteryHelper::stateToString(Solid::Battery::ChargeState state)
{
    switch (state)
    {
    case Solid::Battery::NoCharge:
        return tr("Empty");
    case Solid::Battery::Discharging:
        return tr("Discharging");
    case Solid::Battery::FullyCharged:
        return tr("Fully charged");
    case Solid::Battery::Charging:
    default:
        return tr("Charging");
    }
}
```
whereas the solid doc/comment on the enum says:
```cpp
      /**
       * This enum type defines charge state of a battery
       *
       * - NoCharge : Battery charge is stable, not charging or discharging or
       *              the state is Unknown
       * - Charging : Battery is charging
       * - Discharging : Battery is discharging
       * - FullyCharged: The battery is fully charged; a battery not necessarily
       *                 charges up to 100%
       */
      enum ChargeState { NoCharge, Charging, Discharging, FullyCharged };
```

We should probably revisit the logic and interpretation of the `NoCharge` state in the whole powermanagement code base.

tsujan, 2022-06-15:
I started to see such tooltips. For a few minutes, the tooltip said "Empty (99%)", while the laptop was plugged in. The icon was affected too. I had time only for taking a screenshot:

[image]

I'm sure it didn't happen before; otherwise, I'd notice it. Because of that, I think it may be a new problem in `solid`. Its version is 5.94.0 here, upgraded in May.

tsujan, 2022-06-15:
> Unplug the charger for 5 minutes

In my case, the laptop was plugged in for hours.

"99%" was normal — I'd seen it with other plugged laptops too; it should be related to automatic battery maintenance — but the unplugged icon and the word "Empty" were quite unusual.

tsujan, 2022-06-15:
It's still a mystery to me why I hadn't seen this before, but as @palinek rightly said, `Solid::Battery::NoCharge` doesn't mean "Empty". We should put another word there. I propose "Stable".

That simple change (Empty → Stable) seems enough for solving the problem.

> No emblem for "adapter connected" but it should be displayed.

I think the connection emblem shouldn't be shown in the "Stable" state. We've never been interested in the adapter directly:
 * The circle icons show whether the battery is fully charged, charging, discharging or none of them. There's no assumption about the fourth case in the code, and the "Stable" state is the fourth case.
 * The battery icons are simpler. They only show charging or lack of it, and the "Stable" state is the latter.
 * The theme icons are also simple: charging or lack of it.

So, IMO, we don't need to touch the current icon handling.

closed 2022-08-02 by tsujan