lxqt/lxqt-powermanagement #325: [Feature Request] Disable idle watcher on AC only. Leave enabled on Battery.

Bearded-Blunder, 2022-05-14:
Other power managers (Xfce) allow this.

##### Expected Behavior

Should be able to plain disable interfering with the monitor output on AC, while having it controlled on battery.

A second checkbox for Idle watcher on AC or Idle watcher on battery, not just separate timeouts within a far too restrictive range.  Infinite screen blank timeout should be possible, at **least on** AC, while timeouts to save power are enabled on battery (Laptop **or** UPS).

##### Current Behavior

Currently you have to disable for both or neither. Assuming you're using SystemD otherwise you're SOL & idle watcher settings get completely ignored & short of bash scripts to disable power management of the monitor completely, which defeats the object of this app, you're basically screwed.

##### Desired Outcome

Have the screen remain ON (never blank) on AC, and dim, then power off on battery.
Currently that's not possible, even using the abortion that is SystemD.

Also unless Using SystemD you can't disable the damned idle-watcher AT ALL, it blanks the screen no matter what after the default timeout, regardless what timeout is configured under any other init...

##### System Information
* Distribution & Version: Gentoo (current using openrc + elogind) Debian Stable, and Debian Sid (using sysvInit-core with or without OpenRC as service manager, Debian Stable using runit-init, Current Artix. (Yes I'm a distro-hopper, but less so a DE hopper, I actually ***LIKE*** LXQt, it's my second choice "DE" after the one offered bt BunsenLabs, and like many I DETEST SystemD, though using that hot-mess at least changing the timeout or pausing works)
* Kernel: Varies
* Qt Version: Varies
* liblxqt Version: Varies
* lxqt-build-tools Version: Varies
* Package version: Varies.

Currently about the only "solution" I have is to use xfce4-power-manager in LXQt instead of the native application.. & I'd really prefer not to be using GTK thing in a Qt environment when avoidable. I'm not particularly happy at the prospect of pulling in 3/4 of KDE to use their Qt option (at the expense ~29.9 times the disk usage, I checked,  & lord alone knows what extra resource consumption if I was fool enough to install that & all it's depends).

I fully expect this to get closed, or marked as invalid, because L. Poettering == some Deity.

tsujan, 2022-05-14:
We have separate combo-boxes for AC and battery. Choose "Nothing" with AC.

closed 2022-05-14 by tsujan

Bearded-Blunder, 2022-05-15:
If setting or unsetting those boxes made any difference whatsoever on a non-systemd system, then I wouldn't have posted.  As far as screen blanking on AC goes on such systems, you literally can't do anything from the software that makes a scrap of difference to what happens,

Since my battery is knackered I can't really comment on the "on battery" behaviour, but NOTHING you do in this makes any difference at all on AC, you can bypass the software & disable blanking from the command-line, or you can have blanking after the default timeout, regardless of the state of said combo-box.

tsujan, 2022-05-15:
> As far as screen blanking on AC goes on...

If you want to disable DPMS altogether, see https://wiki.archlinux.org/title/Display_Power_Management_Signaling#Disabling_DPMS