lxqt/lxqt-powermanagement #327: [question] "inhibit screensaver / powersave" mechanism

fenugrec, 2022-06-13:
##### Expected Behavior
Some software (like VLC) have an option such as "inhibit screensaver during playback". It would be nice if lxqt-powermanagement could somehow honour this also for preventing display blanking etc.

##### Current Behavior
Still with the VLC example, 5 minutes into a movie my display will turn off, as configured in lxqt-powermanagement . The option to "inhibit screensaver" has no effect. Same thing in other software.

##### Possible Solution
I'm not sure. It seems most software do something like `xdg-screensaver suspend` / resume , but in my case I don't even run a xdg-screensaver daemon.

##### Context
Perhaps I have something configured wrong, but it's not obvious since everything else works as expected. If this is out of scope for this project, any suggestions are welcome. Thanks !

##### System Information
* Distribution & Version:  arch linux
* Kernel:  5.18
* Qt Version: 6.3.0
* liblxqt Version: 1.1.0
* lxqt-build-tools Version:  n/a
* Package version:  1.1.0

tsujan, 2022-06-13:
It's VLC that doesn't do the job that it's supposed to do; you report it here and not to VLC?!

closed 2022-06-13 by tsujan

fenugrec, 2022-06-13:
Well shouldn't lxqt provide something like org.freedesktop.ScreenSaver so that standard methods like ` xdg-screensaver suspend`can work ? I thought the idea was to try and standardize that kind of behaviour

tsujan, 2022-06-13:
This isn't related to LXQt at all.

Even a basic Qt code like this can disable power-saving under X11 (copy-pasted from a video player):
```c++
  // disable screensaver by shaking the mouse cursor every minute
  QTimer *disableScreenSaver = new QTimer(this);
  connect(disableScreenSaver, &QTimer::timeout, [this] {
    if (isVideo_ && isPlaying_ && isActiveWindow()) {
      QPoint curPos = QCursor::pos();
      QCursor::setPos(QCursor::pos() + QPoint(1, 0));
      if (curPos == QCursor::pos()) {
        QCursor::setPos(QCursor::pos() + QPoint(-1, 0));
        QCursor::setPos(QCursor::pos() + QPoint(1, 0));
      }
      else
        QCursor::setPos(QCursor::pos() + QPoint(-1, 0));
    }
  });
  disableScreenSaver->start(60000);
```
Either VLC doesn't have any code for preventing idleness or its code is inefficient. SMPlayer efficiently disables power-saving with full-screen videos (and with a code much more complex than the above).

fenugrec, 2022-06-13:
Thanks for the example and references. Will look into this

stefonarch, 2022-06-14:
Just for the record: the setting "disable idleness checks when fullscreen" works fine here with VLC.

tsujan, 2022-06-14:
> the setting "disable idleness checks when fullscreen" works fine here with VLC.

Yes, it works independently of the app.