lxqt/lxqt-powermanagement #335: "QString::arg: Argument missing" in journalctl

stefonarch, 2022-08-11:
Not sure if that is an issue but lxqt-powermanagement is spamming the log with the remaining battery time:
```
$ journalctl -xe
....
ago 11 22:25:02 vito lxqt-powermanagement[32269]: QString::arg: Argument missing: "21 minuti" , 21
ago 11 22:25:02 vito lxqt-powermanagement[32269]: QString::arg: Argument missing: "2 ore" , 2
ago 11 22:25:02 vito lxqt-powermanagement[32269]: BatteryChanged discharging: true chargeLevel: 79.4521 actionTim>
ago 11 22:25:32 vito lxqt-powermanagement[32269]: QString::arg: Argument missing: "21 minuti" , 21
ago 11 22:25:32 vito lxqt-powermanagement[32269]: QString::arg: Argument missing: "2 ore" , 2
ago 11 22:25:32 vito lxqt-powermanagement[32269]: BatteryChanged discharging: true chargeLevel: 79.1227 actionTim>
ago 11 22:26:02 vito lxqt-powermanagement[32269]: QString::arg: Argument missing: "14 minuti" , 14
ago 11 22:26:02 vito lxqt-powermanagement[32269]: QString::arg: Argument missing: "2 ore" , 2
ago 11 22:26:02 vito lxqt-powermanagement[32269]: QString::arg: Argument missing: "14 minuti" , 14
ago 11 22:26:02 vito lxqt-powermanagement[32269]: QString::arg: Argument missing: "2 ore" , 2

```

##### Steps to Reproduce (for bugs)

1. Unplug power supply
2. run` journalctl -xe`

* Distribution & Version: latest git, arch

tsujan, 2022-08-11:
I renamed `~/.config/QtProject/qtlogging.ini` to see all Qt warnings, but there was none when I stopped `lxqt-powermanagement` and restarted it in QTerminal. The log was normal:

```
BatteryChanged discharging: false chargeLevel: 98.3112 actionTime: QTime(Invalid)
Inhibit got: 34
Starting idlenesswatcher
"PowerButton" "" ""
"SuspendButton" "" ""
"HibenateButton" "" ""
```

The arguments seems OK in the code too.

tsujan, 2022-08-11:
Oh, I forgot to unplug the power. Yes, there are Qt warnings:

```
QString::arg: Argument missing: "1 minute(s)" , 1
QString::arg: Argument missing: "20 hour(s)" , 20
BatteryChanged discharging: true chargeLevel: 98.3112 actionTime: QTime(Invalid)
Inhibit got: 34
Starting idlenesswatcher
"PowerButton" "" ""
"SuspendButton" "" ""
"HibenateButton" "" ""
```