lxqt/lxqt-powermanagement #338: lxqt-powermanagement v1.1 crashes after waking from suspend

thedoctar, 2022-08-25:
After waking from suspend, lxqt-powermanagment v1.1 crashes. I also get error messages about xscreensaver crashing; unsure if related. Here is some crash info:

`lxqt-powermanagement crashed with SIGSEGV in QHashData::nextNode()`

```
== SegvAnalysis =================================
Segfault happened at: 0x7fa417fc1017 <_ZN9QHashData8nextNodeEPNS_4NodeE+7>:     cmpq   $0x0,(%r8)
PC (0x7fa417fc1017) ok
source "$0x0" ok
destination "(%r8)" (0x5552a8253a6f) not located in a known VMA region (needed writable region)!

== SegvReason =================================
writing unknown VMA
== Stacktrace =================================
#0  0x00007fa417fc1017 in QHashData::nextNode(QHashData::Node*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007fa41935eee8 in  () at /lib/x86_64-linux-gnu/libKF5IdleTime.so.5
#2  0x00007fa4181aa783 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007fa41935ec75 in AbstractSystemPoller::timeoutReached(int) () at /lib/x86_64-linux-gnu/libKF5IdleTime.so.5
#4  0x00007fa4125c9e3b in  () at /usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/org.kde.kidletime.platforms/KF5IdleTimeXcbPlugin0.so
#5  0x00007fa4181701f7 in QAbstractEventDispatcher::filterNativeEvent(QByteArray const&, void*, long*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007fa413c85b05 in QXcbConnection::handleXcbEvent(xcb_generic_event_t*) () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#7  0x00007fa413c87116 in QXcbConnection::processXcbEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#8  0x00007fa413cadd67 in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#9  0x00007fa417617d1b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007fa41766c6f8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007fa4176153c3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007fa4181cc11e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007fa41817174b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007fa418179ce4 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00005557fcb86764 in  ()
#16 0x00007fa4179a5d90 in __libc_start_call_main (main=main@entry=0x5557fcb864b0, argc=argc@entry=1, argv=argv@entry=0x7ffca6912878) at ../sysdeps/nptl/libc_start_call_main.h:58
        self = <optimised out>
        result = <optimised out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {0, -6543805407985233540, 140723103017080, 93836390458544, 93836390565368, 140342775767104, 6545690158029640060, 6514338742198112636}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x5557fcba2008, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = -54910968}}}
        not_first_call = <optimised out>
#17 0x00007fa4179a5e40 in __libc_start_main_impl (main=0x5557fcb864b0, argc=1, argv=0x7ffca6912878, init=<optimised out>, fini=<optimised out>, rtld_fini=<optimised out>, stack_end=0x7ffca6912868) at ../csu/libc-start.c:392
#18 0x00005557fcb87095 in _start ()

== StacktraceAddressSignature =================================
/usr/bin/lxqt-powermanagement:11:/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3+7a017:/usr/lib/x86_64-linux-gnu/libKF5IdleTime.so.5.92.0+ee8:/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3+263783:/usr/lib/x86_64-linux-gnu/libKF5IdleTime.so.5.92.0+c75:/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/org.kde.kidletime.platforms/KF5IdleTimeXcbPlugin0.so+1e3b:/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3+2291f7:/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.15.3+12b05:/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.15.3+14116:/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.15.3+3ad67:/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.1+38d1b:/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.1+8d6f8:/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.1+363c3:/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3+28511e:/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3+22a74b:/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3+232ce4

== StacktraceTop =================================
QHashData::nextNode(QHashData::Node*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
() at /lib/x86_64-linux-gnu/libKF5IdleTime.so.5
() at /lib/x86_64-linux-gnu/libQt5Core.so.5
AbstractSystemPoller::timeoutReached(int) () at /lib/x86_64-linux-gnu/libKF5IdleTime.so.5
() at /usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/org.kde.kidletime.platforms/KF5IdleTimeXcbPlugin0.so
== ThreadStacktrace =================================
.
Thread 4 (Thread 0x7fa4125a8640 (LWP 1183)):
#0  pselect64_syscall (sigmask=<optimised out>, timeout=<optimised out>, exceptfds=0x0, writefds=0x0, readfds=0x7fa41259fac0, nfds=18) at ../sysdeps/unix/sysv/linux/pselect.c:34
        sc_ret = -514
        sc_cancel_oldtype = 0
        data = {0, 8}
        tval = {tv_sec = 0, tv_nsec = 517517378}
#1  __pselect (nfds=18, readfds=0x7fa41259fac0, writefds=0x0, exceptfds=0x0, timeout=<optimised out>, sigmask=<optimised out>) at ../sysdeps/unix/sysv/linux/pselect.c:56
        tval = {tv_sec = 0, tv_nsec = 517517378}
#2  0x00007fa41655903d in  () at /lib/x86_64-linux-gnu/libusbmuxd-2.0.so.6
#3  0x00007fa416559d88 in  () at /lib/x86_64-linux-gnu/libusbmuxd-2.0.so.6
#4  0x00007fa417a10b43 in start_thread (arg=<optimised out>) at ./nptl/pthread_create.c:442
        ret = <optimised out>
        pd = <optimised out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140723103014656, -6543805407985364612, 140342659286592, 2, 140342747793488, 140723103015008, 6514350285958189436, 6514339097203609980}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimised out>
#5  0x00007fa417aa2a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
.
Thread 3 (Thread 0x7fa4130c2640 (LWP 1178)):
#0  0x00007fa417a94d7f in __GI___poll (fds=0x7fa404026f80, nfds=6, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
        sc_ret = -516
        sc_cancel_oldtype = 0
[... 29 lines cut ...]
Thread 1 (Thread 0x7fa414343940 (LWP 1169)):
#0  0x00007fa417fc1017 in QHashData::nextNode(QHashData::Node*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007fa41935eee8 in  () at /lib/x86_64-linux-gnu/libKF5IdleTime.so.5
#2  0x00007fa4181aa783 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007fa41935ec75 in AbstractSystemPoller::timeoutReached(int) () at /lib/x86_64-linux-gnu/libKF5IdleTime.so.5
#4  0x00007fa4125c9e3b in  () at /usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/org.kde.kidletime.platforms/KF5IdleTimeXcbPlugin0.so
#5  0x00007fa4181701f7 in QAbstractEventDispatcher::filterNativeEvent(QByteArray const&, void*, long*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007fa413c85b05 in QXcbConnection::handleXcbEvent(xcb_generic_event_t*) () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#7  0x00007fa413c87116 in QXcbConnection::processXcbEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#8  0x00007fa413cadd67 in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#9  0x00007fa417617d1b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007fa41766c6f8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007fa4176153c3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007fa4181cc11e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007fa41817174b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007fa418179ce4 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00005557fcb86764 in  ()
#16 0x00007fa4179a5d90 in __libc_start_call_main (main=main@entry=0x5557fcb864b0, argc=argc@entry=1, argv=argv@entry=0x7ffca6912878) at ../sysdeps/nptl/libc_start_call_main.h:58
        self = <optimised out>
        result = <optimised out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {0, -6543805407985233540, 140723103017080, 93836390458544, 93836390565368, 140342775767104, 6545690158029640060, 6514338742198112636}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x5557fcba2008, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = -54910968}}}
        not_first_call = <optimised out>
#17 0x00007fa4179a5e40 in __libc_start_main_impl (main=0x5557fcb864b0, argc=1, argv=0x7ffca6912878, init=<optimised out>, fini=<optimised out>, rtld_fini=<optimised out>, stack_end=0x7ffca6912868) at ../csu/libc-start.c:392
#18 0x00005557fcb87095 in _start ()
```

stefonarch, 2022-08-25:
Please complete the [ issue template](https://github.com/lxqt/lxqt-powermanagement/issues/new).

tsujan, 2022-08-25:
The trace show that the crash is in KIdleTime, not in `lxqt-powermanagement`.

tsujan, 2022-08-25:
Arch's is 5.97.0 now. I've had no problem with any version.

stefonarch, 2023-03-18:
Probably outdated  now.

closed 2023-03-18 by stefonarch

tsujan, 2023-03-18:
> Probably outdated now.

Why? doesn't it crash on Wayland? We haven't done anything about it.

tsujan, 2023-03-18:
Oh, sorry, this isn't about Wayland! I had mistaken it with your report.