lxqt/lxqt-powermanagement #354: Idleness Watcher activates despite having Caffeine enabled

Daenarii, 2023-01-14:
LXQt still does its idleness check despite preventing it through the Caffeine program or having a game running (eg. Minecraft).

##### Expected Behavior

It should not suspend (the option I set) when idle. I tried it on KDE Plasma and having Minecraft running doesn't make the system "idle" and also when enabling Plasma's Power Management in the Battery & Brightness system tray option.

##### Current Behavior

Treats the system as idle despite having those programs running.

##### Possible Solution

I think LXQt ignores those programs and only looks at system activity (mouse movement and keyboard use). Maybe having such programs checked will fix it.

##### Steps to Reproduce (for bugs)

1. Run Caffeine
2. Enable Caffeine
3. Let the system idle
4. The system sleeps no matter what

##### Context

I am automating things inside a game and I want it to stay awake. I don't want to fully disable the setting or toggle it every time I want it disabled though as it is annoying and I sometimes forget to do.

##### System Information

* Distribution & Version: Arch Linux
* Kernel: 5.15.87-1-lts
* Qt Version: 1.2.0
* liblxqt Version: 1.2.0-1
* lxqt-build-tools Version: n/a
* Package version: 1.2.0-1

swaggboi, 2023-07-11:
I'm having this issue too running LXQt 1.2.0/Debian 12/Linux 6.1.0-10-amd64. One thing I noticed is that caffeine will prevent XScreenSaver from kicking in but it still does _not_ prevent `lxqt-powermanagement` from detecting 'idle' and taking whatever action you have configured, I'm not sure what's up with that

tsujan, 2023-07-11:
lxqt-powermanagement uses `kidletime` to monitor user activity. It's the job an app like Kaffeine to break idleness correctly if needed. lxqt-powermanagement doesn't and shouldn't work based on an app list.

Kaffeine's code is apparently too old. In its `MediaWidget::checkScreenSaver`, it checks `"org.freedesktop.ScreenSaver` and `org.gnome.ScreenSaver` and then calls `XScreenSaverSuspend()`, instead of using [`KIdleTime::simulateUserActivity`](https://api.kde.org/frameworks/kidletime/html/classKIdleTime.html#a66634c763ed38f915ac7ec96747ef7df). The irony is that both Kaffeine and `kidletime` are made by KDE team.

tsujan, 2023-07-11:
Oh, sorry, Caffeine and Kaffeine have no relation to each other. However, the same argument holds: lxqt-powermanagement uses `kidletime`.

closed 2023-07-11 by tsujan

swaggboi, 2023-07-11:
Thank you for the explanation. I didn't realize this but if you add the power management icon to a panel then you can right-click that and pause the idleness checks for a period of time. This doesn't prevent XScreenSaver but does prevent the idle actions so using them together accomplishes what I was trying to do.

Daenarii, 2023-08-18:
> lxqt-powermanagement uses kidletime
So what's the solution to this? Is it something I should report to caffeine devs?

However if possible, can you add a toggle on the panel to disable idleness check? Or maybe a longer/custom duration so I can leave it overnight for example. Since 4 hours doesn't look enough

tsujan, 2023-08-18:
An app that needs to prevent idleness *everywhere* should do so independently of the DE, e.g., as MPV does. The apps you mentioned may target KDE, GNOME and some other DEs, without taking a general, DE-agnostic approach.

> Or maybe a longer/custom duration so I can leave it overnight for example.

Power Manager Settings → Idle → Change combo-boxes to "Nothing" .

Daenarii, 2023-08-19:
So is it Caffeine's fault that LXQt still sleeps even though Caffeine is activated? Should I make a bug report on their repo?

stefonarch, 2023-08-19:
As the icon for powermanagement is not shown if there is no battery I have a button in the "quicklaunch" widget on the panel which suspends idleness checks calling this script here:

```

#!/bin/bash
path_file="/home/user/.config/lxqt/lxqt-powermanagement.conf"

if grep -q 'enableIdlenessWatcher=false' "$path_file"
        then
        sed -i 's/enableIdlenessWatcher=false/enableIdlenessWatcher=true/' "$path_file"
        notify-send -t 4800 -i user-idle -a Powermanagement "Idleness check enabled"
else
        sed -i 's/enableIdlenessWatcher=true/enableIdlenessWatcher=false/' "$path_file"
        notify-send -t 4800 -a  Powermanagement -i user-idle "Idleness check disabled"
fi

```

Probably it is possible to write a script that checks if kaffeine is running every some minutes and changes idleness settings.

Daenarii, 2023-08-19:
I made a customcommand widget with this as the command to run:

`if [[ $(grep "enableIdlenessWatcher=true" /home/rinna/.config/lxqt/lxqt-powermanagement.conf) ]] ; then echo ON; else echo OFF; fi`

And your script saved in a .sh file as the mouse command when clicked.
It will show the ON/OFF status and clicking it will toggle it

Here's what the widget settings looks like
[image]

stefonarch, 2023-08-19:
Very nice solution! Consider posting this here too: https://github.com/lxqt/lxqt/discussions/2179
- the custom command plugin is one of the most useful widgets IMO.

tsujan, 2023-08-19:
> Should I make a bug report on their repo?

If Caffeine is supposed to prevent idleness (like MPV or SMPlayer does), yes, you could do so.

tsujan, 2023-08-19:
> the custom command plugin is one of the most useful widgets IMO.

@stefonarch, any news of its author? If I remember correctly, he also made "Wing Menu" but didn't make a PR. Any chance to ask him to do so?

Daenarii, 2023-08-19:
> If Caffeine is supposed to prevent idleness (like MPV or SMPlayer does), yes, you could do so.

It's not a media player btw, it's just a program that stays in tray which emulates like an idle application, and I'm specifically using [caffeine-ng](https://codeberg.org/WhyNotHugo/caffeine-ng/). (I think the original caffeine is different now and I can't seem to find it)
I have made an issue on their git repo hopefully they figure out how to do it on LXQt. Thanks for all these information.

tsujan, 2023-08-19:
> It's not a media player btw

Right. I realized that at https://github.com/lxqt/lxqt-powermanagement/issues/354#issuecomment-1630264710.

> hopefully they figure out how to do it on LXQt

LXQt isn't special in this regard. **If** their app is supposed to prevent idleness (I have no idea whether it is), it might need to do it in a DE-agnostic way.

I mentioned MPV not because I thought caffeine was a media player, but because MPV not only does it in all X11-based DEs but also under Wayland. Doing it for X11 may be enough though.

stefonarch, 2023-08-19:
> > the custom command plugin is one of the most useful widgets IMO.
> 
> @stefonarch, any news of its author? If I remember correctly, he also made "Wing Menu" but didn't make a PR. Any chance to ask him to do so?

No sign,  no activity to find searching for "elviosak", kind of algaida story or worse.