lxqt/lxqt-powermanagement #360: [Feature request] Show Remaining Time (again)

ceever, 2023-02-19:
##### Context
I know this has already been discussed before under #175. But since it is such a useful feature also available with Windows, I would like to propose to just use the information about the remaining time (for charging or discharging) from *upower*:
```bash
upower -i $(upower -e | grep '/battery') | grep --color=never -E "to\ full|to\ empty"
```

I actually currently have a keyboard shortcut defined, which extracts the relevant information from *upower* and then triggers a notification with the relevant information displayed. 🤷

[image]

##### Expected Behavior
1. Show remaining time (for charging or discharging) when hovering over the battery icon in the panel with the mouse.
2. Add an option to Powermanagement to have this time displayed on mouseover.

##### Current Behavior
Neither.

##### Possible Solution
Leverage *upower*.

##### Steps to Reproduce (for bugs)
1. Hover over the battery icon shown in the LXQt panel. 

##### System Information

* Distribution & Version: Ubuntu 22.10
* Kernel: lubuntu 5.19.0-21-generic
* Qt Version: 5.15.6
* liblxqt Version: ?
* lxqt-build-tools Version: ?
* Package version: ?

tsujan, 2023-02-19:
You mean something other than this tooltip?

[image]

ceever, 2023-02-19:
Ups ... 😅 ... this wasn't just there when I checked it.

My bad. Thx 🙏

closed 2023-02-19 by ceever

tsujan, 2023-02-19:
Frankly, I don't remember how `solid` worked 4 years ago (when I closed the other report). Anyway, it gives both time-to-empty and time-to-full, and they're shown on the tooltip.