lxqt/lxqt-powermanagement #362: Crash when logging in while laptop battery is unplugged

glibg10b, 2023-03-12:
##### Expected Behavior

No crash

##### Current Behavior

The LXQt Power Management module crashes on login if no battery is connected. It doesn't crash when disconnecting the battery while LXQt is already running

Reconnecting the battery doesn't restart the service, so I have to start it manually

##### Steps to Reproduce (for bugs)

1. Plug in your laptop
2. Disconnect/remove your laptop's battery (some laptops require pushing a pin into a hole to disconnect the battery)
3. Start LXQt

##### Context

I remove the battery when using my laptop at home to increase the battery's lifespan

##### System Information

* Distribution & Version: Gentoo Linux 2.9
* Kernel: 6.1.12
* Qt Version: Too lazy to check. Unlikely to be relevant
* liblxqt Version: 1.2.0
* lxqt-build-tools Version: 0.12.0
* Package version: lxqt-powermanagement-1.2.0

tsujan, 2023-03-12:
Could you provide a backtrace? It's after `where`:
```
coredumpctl gdb ...
where
```

tsujan, 2024-11-04:
No reply. Closing...

closed 2024-11-04 by tsujan