lxqt/lxqt-powermanagement #374: lxqt-powermanagement should override DPMS (under X11)

tsujan, 2023-07-19:
Default DPMS timeouts are in conflict with `lxqt-powermanagement`.

Users might have found various ways of avoiding this problem (like I did), and once the issue was avoided, it might have been forgotten. But, IMHO, it's a serious issue under X11.

Its fix may be as simple as this (inside `IdlenessWatcher::setup()`):

```c++
        // override DPMS settings
        if (QGuiApplication::platformName() == QStringLiteral("xcb")) {
            xcb_dpms_set_timeouts(QX11Info::connection(), 0, 0, 0);
        }
```

This solution worked here. Now `xset q` shows

```
DPMS (Display Power Management Signaling):
  Standby: 0    Suspend: 0    Off: 0
  DPMS is Enabled
  Monitor is On
```

However, not being an expert in cmake, I had a hard time to find XCB's DPMS module with it. In the end, I edited what I found in screengrab.

@luis-pereira, @palinek, @yan12125 
Am I missing something when I think that this issue is serious? And do you know of a simpler way of finding DPMS? I mean simpler than what's done in screengrab?

palinek, 2023-07-19:
> Am I missing something when I think that this issue is serious? 

I, for one, am disabling the DPMS in a startup script.

> And do you know of a simpler way of finding DPMS?

I don't know, what's in screengrab. But there is a find module in lxqt-build-tools for xcb (which is just a boilerplate for pkgconfig information provided by the xcb libs) and with that it should be easy as
```
find_package(XCB REQUIRED COMPONENTS xcb-dpms)
```

> Its fix may be as simple as this (inside IdlenessWatcher::setup()):

I would say, that lxqt-powermanagement should not silently touch other components settings, but be clear about it, i.e.
- adding a config flag, if it should override the DPMS setting while running
- restore the original value upon finishing

tsujan, 2023-07-19:
> I, for one, am disabling the DPMS in a startup script.

Like me.

> But there is a find module in lxqt-build-tools for xcb

Oh, thanks a lot! It hadn't occurred to me to use `lxqt-build-tools` here. Yes, it should solve the problem. And it could be used in screengrab too.

> I would say, that lxqt-powermanagement should not silently touch other components settings

It won't (see below). And because of its job, it should do it, as a KDE component, whose name I don't remember now, does it (`powerdevil`?).

> adding a config flag

It already exists: Enable Idleness Watcher.

> restore the original value upon finishing

That's logical. If I don't find the needed functions easily (because the existing documentations don't seem to explain anything but just add obvious descriptions), I might make a simple PR for other devs to complete it and implement your last criterion.

tsujan, 2023-07-22:
Reopening. See https://github.com/lxqt/lxqt-powermanagement/pull/375#issuecomment-1646550309 for the reason.

closed 2023-08-02 by tsujan