lxqt/lxqt-powermanagement #381: Button settings do not work on wayland

stefonarch, 2023-08-20:
Under wayland power button (and probably suspend|hibernate buttons too) cannot be configured for actions.

##### Steps to Reproduce (for bugs)

1.  Configure an action for powerbutton in `lxqt-config-powermanagement`
2. Press the button
3. Nothing happens

                      
* Distribution & Version: all, git

stefonarch, 2023-08-20:
In the meantime compositors config file can be used, for example in labwc

```
    <keybind key="XF86_PowerOff">
      <action name="Execute">
        <command>wlogout</command>
      </action>
    </keybind>
```

tsujan, 2023-08-20:
If I correctly remember, `lxqt-powermanagement` needed `lxqt-globalkeys` for this, and I clearly remember that the file `lxqt-globalkeys` → `daemon/core.cpp` was for X11.

stefonarch, 2023-08-20:
If `lxqt-globalkeysd` can register shortcuts it will be solved probably so? We'll end up with the same situation as in X11: compositor and LXQt both registering shortcuts which could be contradicting.

I wonder if a "wrapper" approach would be possible, e.g. writing the shortcuts also to the compositors config files and reload its config -  which with labwc would be easy.

tsujan, 2023-08-20:
> compositor and LXQt both registering shortcuts which could be contradicting.

Yes.

> I wonder if a "wrapper" approach would be possible, e.g. writing the shortcuts also to the compositors config files

No, for two reasons at least: that'll be *very* ugly if possible; and we'll need our own shortcut manager, as we do with X11.

IMO, a WM shouldn't handle shortcuts — after all, it's a *window* manager — but a separate daemon should. I think shortcut handling through WM may be an old habit

stefonarch, 2024-06-07:
I'm wondering what we should do here atm:
[image]

Greying out under wayland makes no sense IMO. 
We could add (X11) to the the first line and a text box at the bottom about configuring  XF86PowerOff, XF86Hibernate and XF86Sleep in the compositor, but maybe this is too tecnical.

tsujan, 2024-06-07:
> Greying out under wayland makes no sense IMO.

That may be our only option for now.

stefonarch, 2024-06-07:
I don't think this is logical: it is a setting, it can be changed but it affects X11 only. 
And we don't grey out neither WM setting or x11 screenlocker for x11 under wayland.

Mockup:
[image]

tsujan, 2024-06-07:
OK, but it may be better to leave it as it is under X11 and show the extra message only under Wayland:

```
Note: Under Wayland....
```

stefonarch, 2024-06-07:
That would be perfect but how can we change an .ui file under wayland? Or a popup when opening this part?

tsujan, 2024-06-07:
It can be done in the code. Popups are annoying unless really needed. If we agree on this, I'll do it.

stefonarch, 2024-06-07:
I'm ok with that, nice solution not to bother x11 session settings with that.

closed 2024-06-08 by tsujan