lxqt/lxqt-powermanagement #383: Shows "remaining 0 minutes" / "to full 0 minutes" after charging state change

GTcreyon, 2023-10-20:
##### Expected Behavior

Theoretically, a reasonable estimation would be displayed immediately. I would assume that this is impossible in practice, due to needing time to gather data with which to make said estimate, so an alternative would be, perhaps, "to full N/A", "to full unknown", "calculating estimate..."

##### Current Behavior

When a charge cable is connected or disconnected, the estimated time to full/depletion becomes 0 minutes for a short time, up until the next update. This is unusual, because it would imply that the battery will become full/empty instantly.

##### Possible Solution

A divide-by-zero check (if one does not already exist) that returns a handleable exception that can be used to display a more sane estimate rather than 0.

##### Steps to Reproduce (for bugs)

1. Unplug charge cable
2. Hover over the icon
3. Observe the depletion ETA

or,

1. Plug in charge cable
2. Hover over the icon
3. Observe the charging ETA

##### Context

It doesn't seem to be a particularly significant issue. It's relatively clear to a user that this is not a reasonable estimate. I can only imagine it being misleading when the system is already on low battery, and a user has to, say, unplug and run to a different room for whatever reason. In that case, "remaining 0 minutes" might be believable enough to have an impact.

##### System Information

* Distribution & Version: Arch
* Kernel: 6.5.1.arch1-1
* Qt Version: 5.15.11
* liblxqt Version: 1.3.0-2
* lxqt-build-tools Version: None?
* Package version: 1.3.0-2

CladisticFish, 2024-11-04:
I haven't noticed the above problem on Lubuntu 22.04; however, I'm seeing something v similar where I've just installed Lubuntu 24.04 on an HP Stream netbook.

Behaviour: Shows estimated "to full minutes" when charging, but when discharging shows "remaining 0 minute(s)" almost all the time. (Not just for a few minutes after charging state change; did at one point show 30 minutes. TBC, battery life in Windows is approx 11 hours according to reviews, and I have no reason to think it noticeably worse with Lubuntu.) I've only been running Lubuntu on it for the past four days, but it has been partially discharged and charged back up to 100% a couple of times over that period. I'm currently seeing tooltip status "Discharging (84%), remaining 0 minute(s)".

System Information

    Distribution & Version: Lubuntu 24.04
    Kernel: 6.8.0-48-generic (x86_64)
    Qt Version: 5.15.13
    liblxqt Version: 1.4.0-0
    lxqt-build-tools Version: None?
    Package version: 1.4.0-0ubuntu5

tsujan, 2024-11-04:
Not reproducible.

LXQt uses [`Solid::Battery::timeToEmpty()`](https://api.kde.org/frameworks/solid/html/classSolid_1_1Battery.html#a4e2438032ef7848f004236dbe31d907e). Make sure you have an up-to-date KF6 `solid` package. The latest stable version is 6.7.0.

tsujan, 2024-11-04:
> liblxqt Version: 1.4.0-0

That's an old version and not supported by LXQt. Please report the issue to Lubuntu.

closed 2024-11-04 by tsujan