lxqt/lxqt-powermanagement #408: No notifications for low/critical battery

Svalorzen, 2024-08-09:
##### Expected Behavior

I'd like to receive a notification on low/critical battery. I tried looking at the docs and at the `/etc/UPower` config file, but I couldn't find anything about it.

##### Current Behavior

This is not happening at the moment. I have found [this issue](https://github.com/lxqt/lxqt/issues/673) on the main lxqt repository, but I'm not sure whether it's current or not. The documentation does talk about notifications for missing battery, so it seems strange that they would still be missing for low/critical battery (also because I think I did receive some unrelated notification from powermanagement at some point). I'm sure I did no receive a notification about this as I checked the past notifications received.

If important, I am on a thinkpad with a double battery (one removable, one fixed). I am not sure whether this may affect the low/critical calculations.

##### Possible Solution

##### Steps to Reproduce (for bugs)

I just let my laptop discharge. I am opening this because just a moment ago I had reached 1% & 2% on both my batteries without noticing and not getting any warning.

##### Context

Notifications about low battery are fairly straightforward.

##### System Information

* Distribution & Version: Lubuntu 24.04 
* Kernel: 5.15.0-117
* LXQt Version | 1.4.0
* Qt | 5.15.13
* Build type | Release 
* liblxqt Version: liblxqt1 - 1.4.0-0ubuntu3
* lxqt-build-tools Version: not installed?
* Package version: lxqt-powermanagement - 1.4.0-0ubuntu5

tsujan, 2024-08-09:
A notification is shown when the charge level is below the value you set:

[image]

Please check whether you've enabled the battery watcher.

closed 2024-08-09 by tsujan

Svalorzen, 2024-08-09:
I did check this, my battery watcher is enabled:
[image]

tsujan, 2024-08-09:
You have "Nothing" there, which means what it says.

Svalorzen, 2024-08-09:
That's the action taken on low power, it's not about the notification is it? The options are all about asking what to do, turning off the screen or shutting down. Isn't there supposed to be notifications at low and critical levels?

stefonarch, 2024-08-09:
I'm not 100% sure but notifications are only send if any action is selected. Personally I use "Ask" which is also hard to miss. Basically you would like an option "Notify only"?

tsujan, 2024-08-09:
> an option "Notify only"?

LXQt isn't KDE ;) The current options are more than enough:

[image]

webghostx, 2024-10-19:
I like to set the shutdown to prevent the battery from being drained too much. But the problem I have is that I don't see the message when I'm busy working.

Wouldn't it be an idea to enable an acoustic warning message, or simply be able to insert a bash command at the warning?

stefonarch, 2024-10-19:
> But the problem I have is that I don't see the message when I'm busy working.

Notifications should be seen, that's the point of them. Change position and time?
The battery icon can also show percentage, so one look at it is enough. The tooltip shows remaining time too.

webghostx, 2024-10-19:
Yes, of course, I have set all of that up. 500 second warning, I only noticed it today when the lights went out :)

I have now switched to "Ask". It should work as long as you don't leave the notebook alone. I would still find a line for a command helpful here. Otherwise, the only thing that helps is a parallel running script for individual adjustments.