lxqt/lxqt-powermanagement #420: Problem to set lid closed action to suspend by default.

garbancito13, 2025-01-05:
On my first experiences with lxqt, I closed the lid, and after some time, I
found that my new computer was overheating. That is because the closed lid
configurations were set by default to "nothing" instead of "suspend". Fixing
this will improve energy saving and will also improve experience to new users
without previous experience with lxqt.

##### Expected Behavior
Lid closed action should be on suspend by default.

##### Current Behavior
Shows nothing instead of suspend so the machine overheats.

##### Possible Solution
We put the lxqt-powermanagement.conf with lid closed action, low battery action and idle action on suspend to /usr/share/lxqt/lxqt-powermanagement.conf. Yet somehow only idle changed to suspend. Here is our configuration file:
```

[General]
__userfile__=true
backlightIdleness=0
backlightIdlenessOnBatteryDischarging=true
disableIdlenessWhenFullscreen=false
enableBatteryWatcher=true
enableExtMonLidClosedActions=false
enableIdlenessBacklightWatcher=false
enableIdlenessWatcher=true
enableLidWatcher=true
iconType=1
idlenessACAction=4
idlenessACTime=@Variant(\0\0\0\xf\0\r\xbb\xa0)
idlenessBatteryAction=4
idlenessBatteryTime=@Variant(\0\0\0\xf\0\r\xbb\xa0)
idlenessTime=@Variant(\0\0\0\xf\0\0\x3\xe8)
lidClosedAcAction=4
lidClosedAction=4
lidClosedExtMonAcAction=-1
lidClosedExtMonAction=-1
powerLowAction=-1
powerLowLevel=5
powerLowWarning=30
runCheckLevel=1
showIcon=true

```
##### Steps to Reproduce (for bugs)
1. Place our configuration under /usr/share/lxqt/lxqt-powermanagement.conf
2. ```chown root:root  /usr/share/lxqt/lxqt-powermanagement.conf```
3. Logout
4. Login and check on Preferences > LXQt settings > Power Management
##### Context
On my first use, closed the lid and then, after some time I opened it again and it was overheating and the battery was drained.

##### System Information
* Distribution & Version: Debian Trixie (13)
* Kernel: 6.12.6-amd64
* Qt Version: 6.7.2
* liblxqt Version: 2.1.0
* lxqt-build-tools Version: 2.1.0
* Package version: 2.1.0

stefonarch, 2025-01-05:
I've  no problem with setting it to suspend by default, but overheating is clearly another issue, probably non related to LXQt.

garbancito13, 2025-01-06:
Sorry, I think I haven’t explained myself well. The laptop overheated
because it was closed with a bright screensaver for 2 hours. This is due to
the lid closed action being set to nothing instead of suspend.

Kind regards, Joel.

garbancito13, 2025-01-06:
Additionally, in the default gnome desktop, suspend is fine and it
automatically suspends while in LXQt desktop, the laptop stays on and that
might scare newcomers. So please make it suspend by default.

tsujan, 2025-01-06:
Not reproducible, of course.

Some info is missing.

tsujan, 2025-01-06:
Apart from the description of this report, setting suspension as the default action for lid closing makes sense to me; I don't see any benefit to no default action in this case. Am I missing something? @stefonarch?

stefonarch, 2025-01-06:
As I said: it's not an issue normally but can lead to one like in the case here.
Making suspend the default action is fine to me. We could also set powerbutton action to "ask" if it's not already so.

tsujan, 2025-01-06:
> Making suspend the default action is fine to me.

IMHO, it's much more reasonable than doing nothing. Anyone expects that their laptop should go to suspend on closing its lid; they could change that at any time later if they want.

closed 2025-01-06 by tsujan