lxqt/lxqt-powermanagement #425: [Feature request] Power profiles

libalis, 2025-04-08:
Thank you so much for your incredible work on this project.

LXQt is already packed with so many great features, and I truly appreciate all the effort that’s gone into it.

In recent years, desktop environments like GNOME and KDE Plasma have added power profiles such as battery saver, balanced, and performance modes, typically managed by tools like [power-profiles-daemon](https://gitlab.freedesktop.org/upower/power-profiles-daemon).

It would be fantastic if something similar could be implemented in LXQt.

Having the ability to change power profiles directly from the battery widget, as well as through a settings page, would be a great addition.

This would make the LXQt experience even more powerful.

Thanks again for all your hard work!

libalis, 2025-04-08:
Thank you for your response! I understand that LXQt already has power management settings, but I believe having integrated power profiles within LXQt itself would provide a more streamlined and user-friendly experience. While tools like TLP are great for system-wide management, and the power-profiles-daemon is also system-wide, adding this feature directly into LXQt would make it more accessible and convenient for users who prefer adjusting power profiles on the fly. It would enhance LXQt’s usability by allowing users to switch profiles easily from the battery widget, without relying solely on external tools. This would make the LXQt experience even more complete and accessible.

tsujan, 2025-04-08:
> … without relying solely on external tools.

LXQt tries to remain as agnostic as possible; it doesn't treat any app as "external", unless that app is DE-specific. The user is free to use any window manager, network manager, power manager, text editor, …. I'm afraid, even if we there was plenty of manpower, reinvention of the wheel wouldn't be an option.

libalis, 2025-04-08:
I apologize for not expressing myself clearly. I think it's great that LXQt aims to give users as many choices as possible. However, I would like to see an optional GUI for these three power profiles, similar to what obconf-qt does for openbox. Users who prefer tools like TLP will not be stopped by this, but other users, especially those with less experience, would have a very simple and powerful tool to either make their PC faster or extend battery life.

tsujan, 2025-04-08:
Maybe I'm the one who wasn't explicit enough. I hope the following example is clear; will leave this report to other devs after explaining it.

Suppose that someone has made a Qt-based power management profiler. It's unlikely that it'll become an LXQt app, not because it doesn't meet LXQt standards — it may be a very good and easy-to-use app — but because LXQt has no reason to include it. LXQt might recommend it though.

Unlike in GNOME or perhaps KDE, "Integration" doesn't make much sense with LXQt, because LXQt uses Qt, GLib and some other platform-independent libraries. It can also be said that anything can be "integrated" into LXQt.

Last but not least, LXQt is so by design.

libalis, 2025-04-09:
Thank you for helping me understand your perspective better. While our views remain the same, that's perfectly fine—everyone has the right to their own opinions. This is your project, and you have the final authority here. I just wanted to offer my suggestion, and if the other developers don't have any objections, I’ll be happy to go along with it. Thanks again for all the hard work you're putting into this incredible project.

palinek, 2025-04-09:
@tsujan @libalis Can you have a look into #426?
The implementation would be simple and we would expose the power-profiles only in case the DBus object is available.

tsujan, 2025-04-09:
Oh, thanks!

I don't know if this is what the reporter had in mind because he talked about a daemon (he might tell us later), but it could be a nice enhancement indeed. Will check it after the next release.

palinek, 2025-04-09:
> reporter had in mind because he talked about a daemon

It is exactly using the reported daemon interface. We already rely on `org.freedesktop.UPower` DBus service. This implementation relies on `org.freedesktop.UPower.PowerProfiles` service. So nothing exceptionally new in that regard. I just wanted to know the opinion & optionally get some quick UI review to continue (as now the implementation is just a dirty POC).

libalis, 2025-04-09:
That's exactly what I meant, sorry for not explaining it clearly. Thank you so much! I'll compile and test it today. Just two questions:
1. How can you access this option if there's no battery present?
2. Does it display all available profiles, or are they hardcoded?

palinek, 2025-04-09:
> How can you access this option if there's no battery present?

No option as of now. The same as pausing the idleness.

> Does it display all available profiles, or are they hardcoded?

It displays the list, what the `PowerProfiles` provides.

libalis, 2025-04-09:
> No option as of now. The same as pausing the idleness.

A new widget probably doesn't make sense, would it be possible to replace the battery icon and then only offer these two options?

> It displays the list, what the PowerProfiles provides.

Great work, thank you very much!

stefonarch, 2025-04-09:
> > How can you access this option if there's no battery present?
> 
> No option as of now. The same as pausing the idleness.

Does a powerprofile switch make sense without a  battery?
The lack of a systray icon to access disabling idleness checks without battery is a long missing issue IMO.

libalis, 2025-04-09:
The code compiled without any issues and is working great!
However, I have two points to address:
1. When executing the command powerprofilesctl set power-saver, the GUI does not reflect the change.
2. What are your thoughts on using resources-symbolic as the icon?

libalis, 2025-04-09:
Maybe use it as the icon for the widget when there's no battery?

palinek, 2025-04-09:
> When executing the command powerprofilesctl set power-saver, the GUI does not reflect the change.

Sure. I'm aware of this in the POC code. Will be addressed in full PR.

> What are your thoughts on using resources-symbolic as the icon?

I, personally, wouldn't use something completely out of [xdg spec](https://specifications.freedesktop.org/icon-naming-spec/latest/)

closed 2025-04-11 by libalis