lxqt/lxqt-powermanagement #430: Revamp idle options

RoydZeller, 2025-04-14:
I had problems with screensavers and found https://github.com/lxqt/lxqt-powermanagement/issues/374

Now, I can disable power management to use `xset` again, but then lid watcher stops working. 

The issue here is that LXQT interferes with other screen lockers, but it can't do everything on it's own:
- You can't blank after 5 minutes, and lock after an additional 5 minutes, suspend after another 5 minutes.
- The options are limited to maximum 1 hour.

Before I could have `xset s`/`xss-lock` handle screen blanking and locking and have lxqt do the suspend (though still unideal due to time limit).

stefonarch, 2025-08-29:
Related https://github.com/lxqt/lxqt-powermanagement/issues/430

stefonarch, 2025-08-29:
Actually already reported:
https://github.com/lxqt/lxqt-powermanagement/issues/232
https://github.com/lxqt/lxqt-powermanagement/issues/387