lxqt/lxqt-powermanagement #439: Option "Turn off monitor(s)" not working on Wayland

stefonarch, 2025-07-02:
In `~/.config/lxqt/power.conf` several commands can be used for specific compositors:
```
#kwin
monitorOffCommand=kscreen-doctor --dpms off
#niri
monitorOffCommand=niri msg action power-off-monitors
#hyprland
hyprctl dispatch dpms off

```

Maybe there are others commands too, like `wlopm` which needs wlr-output-power-management-v1 and monitor names.

The working commands could be included, and a text input field added in the GUI for them.
`monitorOffCommand=xset dpms force off` for X11 could be selected automatically.

stefonarch, 2025-07-02:
Maybe "power off monitors" could also be separated from suspend: example power off monitors after 5min, suspend after 10min.

tsujan, 2025-07-02:
It should be done in parallel to changes to liblxqt: `/usr/share/lxqt/power.conf` belongs to the latter.

tsujan, 2025-07-15:
It's basically implemented now; we can't do more. So, I think this can be closed.

Of course we'll add more recommended commands if they appear, but users are free to use commands of their choice.